Yowkees/keyball #18: ボールの有無で2種類のファームを作る

koron, 2021-06-21:
ボールの有無=基板の種類で2種類のファームを作る。

名前は rev1/ball および rev1/noball とする。

それに先立ち `LAYOUT` の整理をしたほうが良い。
これについては別途 issue を作ります。

---

### Discussed in https://github.com/Yowkees/keyball/discussions/17

<div type='discussions-op-text'>

<sup>Originally posted by **koron** June 21, 2021</sup>
いまボールの有無(基板の違い)によらずファームは1種類ですが、これを分けたほうが良いのではないかとのアイデアを得ました。思考の推移は以下の通りです。

* 左右自動判定やボールの有無の自動判定をやりたい
* ボールの有無の自動判定は何らかの方法でできそう (PMWにResetなげてレスポンスを見てみる)
* 左右の自動判定は無理そう

    ProMicroの未使用ピンがない & 異なる基板で未使用マトリックスが異なるため、1種のファームで2種の基板の左右を判定する(4通り!)ほどよい手段がない。ボールの有無を判定したのち、それに応じてチェックする未使用マトリックスを変えることはできるかもだが…複雑であまり好ましいアイデアとはいえない。

* ならいっそPCB毎に別のファームを用意したらどうか(名前の例: `rev1/onball`, `rev1/offball`)

    名前はあとで変更可

* これなら未使用マトリックスで左右判定できる
* ファームウェアのサイズ削減にもつながり一石二鳥
    * offballはPMWのファームがまるまる要らなくなる(大きな効果)
    * onballはトラックボールの有無判定が要らなくなる(小さな効果)
    * offballにだけviaを入れてUSBを繋ぐ(masterにする)ようにすると、onballからviaを消せてファームに余裕ができるのでは?

        VIAはUSBを繋いでるほうだけで良いはず。TRRSの通信内容はマトリクスの転送やRGB LEDのタイミング同期だけでVIA固有のものはない

* 判定に使うマトリクスは onball が `F7, B5` で offball が `F6, B5` を予定
    * これはonballが右手の default キーマップでは MBTN1 と MBTN2 に相当
* `rev1` は引き続き自動判定なし、トラックボール側にUSBを接続する&左手ボールの場合は適切なファームを使う、で使える</div>

koron, 2021-06-21:
@Yowkees 問題なければぼく = koron に assign しておいてください。

Yowkees, 2021-06-21:
@koron Git初心者でお手数おかけします。assignしました！

koron, 2021-06-23:
#23 でボール有用のファームウェアが完了しました。

あとはボール無し用のファームウェア作って…
両者をつなぐ組み合わせの対応をする必要があります。

koron, 2021-06-24:
おおよそめどはたったのでここで残りの問題点を整理

* [ ] single trackball on secondary のケースでマウスカーソルの動きが1/10になってしまう
* [ ] secondary に繋いでいるモノ次第、それに依存して matrix_mask を変更する必要がある
* [ ] RGBLIGHTのアニメーションを増やす
* [ ] VIAを含めたファームを検討する
* [ ] [VUSBに逆流するProMicro](https://zenn.dev/koron/articles/7cdfd5382f9ee3)への対応を入れる(暫定的にやったらキーボードがハングした)

koron, 2021-06-26:
> single trackball on secondary のケースでマウスカーソルの動きが1/10になってしまう

trackball 周りのファームを書き直して整理しなおしました。

> VIAを含めたファームを検討する

rev1/ball/via, rev1/noball/via, rev1/ball/via_Left, rev1/noball/via_Left がコンパイルできることを確認しました。

最新版はココで作業してます。
https://github.com/koron/qmk_firmware/commit/9a3c4825dbd8516658e0003ce2acc6bf1564b839

koron, 2021-06-26:
> RGBLIGHTのアニメーションを増やす

Via無しでギリギリ入った。

Yowkees, 2021-06-27:
ファームウェアの整備、誠にありがとうございます！
最新版のファームを試してみました！USBをどちらに差しても使用できるのは実に爽快です！TRSケーブル内にトラックボール情報が流れてるのか～と見入ってしまいました。
koronさんは現在ご自身のリポジトリで作業されているということで、Keyball46のビルドガイドに掲載する最新ファームウェアのリンク先をkoron/qmk_firmwareとしても大丈夫でしょうか？

それと１点、わたしのキーボードですとレイヤー３への切り替えが機能しないようなのですが、koronさんのものでは問題ないでしょうか？私のmakeや書き込みが悪かったのかもしれませんので、もう少しいろいろ試してみます。

koron, 2021-06-27:
ああ最新のファームウェアは
こちらの Yowkees さんのレポジトリに置きませんか? 
やはり Yowkees さんのハードなんで。

後程方法を考えましょう。DisussionのAnnouncementにZIPで添付するのが良いかもしれません。

koron, 2021-06-27:
> secondary に繋いでいるモノ次第、それに依存して matrix_mask を変更する必要がある

done

> RGBLIGHTのアニメーションを増やす

容量的に VIA + フルRGBLIGHT + トラックボールドライバは不可能なので初版では諦めましょう。

将来的には トラックボール無し側にVIA+ フルRGBLIGHTのファームを、
トラックボール有り側に RGBLIGHT + トラックボールドライバのファームを書いて
トラックボール側にUSBを繋ぐという制限でVIA+RGBLIGHTが同居できるとは思います。

koron, 2021-06-27:
> VUSBに逆流するProMicroへの対応を入れる(暫定的にやったらキーボードがハングした)

試したところ待ち時間を200msくらいにするとハングした。
正確には secondary と判断されてUSBキーボードとして機能しなくなる。
手元では300msにすれば問題なく動くが、安全マージンを見て500msにしておく。

Yowkees, 2021-06-27:
最新ファームウェアの位置了解しました。お送りいただければこちらで管理いたします。

koron, 2021-06-27:
> わたしのキーボードですとレイヤー３への切り替えが機能しないようなのですが、koronさんのものでは問題ないでしょうか？

いま default キーマップと via で試してみましたが、問題なく _BALL レイヤーに切り替えられました。
一応これで作業がひと段落したので、いまからPR作ります。

closed 2021-06-27 by Yowkees