Yowkees/keyball #19: `LAYOUT` マクロの修正&整理

koron, 2021-06-21:
related https://github.com/Yowkees/keyball/issues/18

今の `LAYOUT` マクロにはいくつか問題があるので修正したほうが良いです。
修正内容は以下の通り。

* 右側ボール、左側ボール、ダブルボール、ノーボール(これいる?)の構成に向けてそれぞれ LAYOUT を用意する

    命名例: `LAYOUT_rightball`, `LAYOUT_leftball`, `LAYOUT_doubleballs`, and `LAYOUT_noballs`

* `LAYOUT` を `LAYOUT_rightball` のエイリアスとする

* 各レイアウトで割当不可能なキー隠す

    例: 右側ボールにおける左の F6, B5 と 右の F7, B5

    handiness の自動判定を考慮するとあった方が良い。

* `LAYOUT_kc` の廃止

    `MO(1)` などを記述するのにわざわざ `KC_*` なキーコードを定義する必要があり、QMKの流儀とは著しく異なる。多少手間でも `KC_Q` などはそのまま書いておいた方が良い。qmk c2json や json2c のことを考えるとあまり良いアイデアとは言えない。

    * 既存の `LAYOUT_kc` を使ってるキーマップを前述の `LAYOUT_rightball` もしくは `LAYOUT` を使ったものに置き換える

koron, 2021-06-21:
@Yowkees こちら、問題なければわたし=koronにassignしておいてください。

koron, 2021-06-21:
今の `LAYOUT_kc` そのものは残しておくことはできます。
ただ提供しているキーマップでは使わない方が良いでしょう。

Yowkees, 2021-06-21:
ノーボールを選択する稀有な方はいないでしょうが、一応準備しますか？

>多少手間でも KC_Q などはそのまま書いておいた方が良い。
そうなのですね。ご指摘ありがとうございます！勉強になります。

koron, 2021-06-22:
#22 で完了しました。これでダブルトラックボールの本題に取り掛かれます。

closed 2021-06-22 by Yowkees