Yowkees/keyball #42: Remap Review Result

yoichiro, 2021-07-04:
Hi, thank you for submitting the review request for this keyboard. However, we have rejected the request, because the PRODUCT value does not match the value written in the `config.h` file. For instance, you filled `Keyball64` in to the review request form, but the value in the `config.h` file is [`keyball`](https://github.com/qmk/qmk_firmware/pull/13185/files#diff-82354efd4bb61eec1c4d05d2023e841f7eac6ed3851d6a34174f983b0dfe7bbbR27). You need to fill `keyball` in to the review request form, or you need to change the PRODUCT value in the `config.h` file to `Keyball64`. Then, please resend the review request.

こんにちは、このキーボードについてレビュー申請を送っていただきまして、ありがとうございます。しかし、そのリクエストは既にリジェクトさせていただいております。理由といたしましては、PRODUCT値が `config.h` ファイルに書かれた値と一致していません。具体的には、レビューリクエストフォームに `Keyball64` と記入されていましたが、 `config.h` ファイル内の値は、 [`keyball`](https://github.com/qmk/qmk_firmware/pull/13185/files#diff-82354efd4bb61eec1c4d05d2023e841f7eac6ed3851d6a34174f983b0dfe7bbbR27) でした。レビューリクエストフォームに `keyball` と記載していただくか、もしくは `config.h` ファイル内の PRODUCT 値を `Keyball64` に変更していただく必要がございます。その後、レビューリクエストを再送信していただけますと助かります。

Yowkees, 2021-07-12:
ご連絡ありがとうございます。
丁寧に教えていただいた通りPRODUCT値を修正して、無事にレビュー通ったようです。
すばらしいRemapサービスのご提供、ありがとうございます！

yoichiro, 2021-07-12:
再申請ありがとうございました。閉じます。

closed 2021-07-12 by yoichiro