Yowkees/keyball #61: Firmware does not compile on qmk 0.14.6

freznel10, 2021-09-02:
Hi I bought the keyball. I tried to compile the firmware and it fails on the latest QMK.

Yowkees, 2021-09-02:
Hi. Thank you for buying Keyball46!
I will check it with newest QMK.

freznel10, 2021-09-03:
Thanks. Looking forward to it. I could not use my keyball right now.

Yowkees, 2021-09-03:
If you want just use Keyball46, please use compiled firmware "hex file".
It has been released in English Build Guide's Chapter 10 "Write the regular firmware". below link.
https://github.com/Yowkees/keyball/blob/main/keyball46/doc/rev1/buildguide_en.md

Please download ZIP file that includes compiled firmware, and just write in your ProMicro via QMK Toolbox.

Now, I am fighting against latest QMK_Firmware.
Compiling with QMK_Firmware 0.13.x is fine.
Just Compiling with latest 0.14.x has problems.

Thank you.

Yowkees, 2021-09-03:
This compiled firmware supports "Remap" website as below.
https://remap-keys.app/configure
Changing Keymap is very easy. Please try it.

freznel10, 2021-09-04:
Thank you. Will dowmlnlod. Please let us know if the firmware’s been
ported. :) I wanted to implement combos and the new key overrides for the
QMK.
Also looking forward to the Keyball65:)

Yowkees, 2021-09-04:
Thank you very much for checking my projects. I will do my best:) Thank you.

freznel10, 2021-10-13:
> @freznel10 <https://github.com/freznel10>
>
> Please try current head.
> It works (be able to compile) with QMK 0.14.22 now.

Will do. Thanks!

Yowkees, 2021-10-14:
@koron 
Thank you very much for your replying  in place on me!