Yowkees/keyball #85: Scroll_modeへの入り方について

Yowkees, 2022-01-24:
@koron 様
公開されましたKeyball61のVIA対応ファームウェアを使用してみてのご提案になります。REMAPを使用してみました。
新しいファームウェアでは、"SCRL_MO"が追加され、これを押下中にスクロールモードに入ることを理解しました。

私が以前より使い慣れたキー配列ですと、左手親指キーの外側キーは、HOLDでLayer3＝scrollモードになり、TAPでかな入力になるように使用していました。これと同じキー配置にしようと試したところ、"SCRL_MO"はHOLDに設定できないようです。@yoichiro さんに伺ったところ、HOLDに指定できるキーはQMK_Firmware側で限定されていたような、、との回答でおそらく出来ないようです。

そこで、新設の"SCRL_MO"は残したまま、Layer3に切り替え中にScrollモードとなる仕様を復活させるのはどうでしょうか？
もし、「Layer3とスクロールモードの連動で問題が発生したために"SCRL_MO"を新設した」というような事情があるようでしたら、作戦考えたいと思いますのでお教えいただければ幸いです。

大変お手数ですが、ご検討いただければ幸いです。

koron, 2022-01-24:
そうですね。VIAキーマップでのみLayer 3でscroll modeをenableする仕様を組み込みましょう。

koron, 2022-01-24:
@Yowkees https://github.com/Yowkees/keyball/actions/runs/1739670752

とりあえず `via` キーマップのみ layer 3 に同期して scroll mode をenableするようにしました。
これで問題ないか確認してみてください。

その上で `default` キーマップにも適用するか検討しましょう。

なお `test` と `develop` にはこれらのキーマップの目的から使い道がないので適用しません。

Yowkees, 2022-01-24:
早速のご対応ありがとうございます！
自キ始めた頃の僕のようなReMAP以外は無理っなユーザーへの自由度が上がると思います🙏
よーし！もう完璧な気はしてますが最終確認です！

koron, 2022-01-24:
せっかくなんで Keyball46 に合わせる意味も込めて `default` も入れておきましょう。

closed 2022-01-24 by koron