Yowkees/keyball #90: Keyball46 failed to resume from sleep

koron, 2022-01-30:
https://twitter.com/kaspinMTH/status/1487649434263785472

> windows 10 proを使用しているのですが、スリープモードから復帰した際、keyball46を認識しません。OLEDにkeyballのロゴが表示された状態になります。 TRRSケーブルを抜き差ししたら認識します。
どうすればスリープ復帰した時点でkeyball46を認識しますか？

kaspinning, 2022-01-30:
PC側のUSB3.0端子に直差ししていたケーブルをUSB2.0端子に繋ぎ変えたら解決しました。

PCをスリープモードにすると、keyballの基板LEDは点灯したままで、トラックボール以外のキーを入力するとスリープモードから復帰し、keyballは認識されたまま使えました。

yswallow, 2022-01-30:
ProMicroのみで左手のファームウェアを試したところ、Ball Rightの左手のみだと問題がなく、Ball Leftの左手のみだと問題が発生したので、Keyballとの合わせ技の問題だと思います。マウスの移動情報を送り続けているためだと思っています。

~~仮説が正しければBall Rightでも左右を接続して左手にUSBケーブルを挿すと問題が発生するはずですがいかがでしょうか？（Ball Leftなら 右手のみ→問題なし、左右接続して右手にUSB→問題あり となるはず）~~

↑手元で試します

koron, 2022-09-10:
no one reports how to reproduce this problem.
so we can't fix it now.

i'll close this once.

please open this again, when you have further information.

---

問題を再現する方法を誰も提示できないので、現在のところ修正の見通しがありません。

よって本件はいったん閉じます。

もしも再現に関して追加情報があったら、再度issueを開きましょう。

closed 2022-09-10 by koron