Yowkees/keyball #112: another key is pressed when pressing two keys

koron, 2022-04-16:
Keyball61, without the ball, left side.

when pressing `v` and `a`, `]` is pressed unexpectedly.

Original report:

> 自分のkeyball61はvとaを同時押しすると[が入力されてしまいます。

(He maps  `[` at `]`'s default position)

https://twitter.com/jomatsu_/status/1514968357841543168

koron, 2022-04-16:
We know that Keyball61 has a problem that when pressing specified two keys
(a key on left side of row 3 and a key on right side of row 4)
other key on  row4 and col3 (`]`) is pressed by noise sometime.

koron, 2022-04-16:
But I confirmed that it doesn't happen when using QMK + ProMicro with my Keyball61.
So there may be some conditions to reproduce this problem:

* soldering status
* enviornmental radio wave
* or so

koron, 2022-04-17:
It looks like that pressing three keys make Keyball61 unstable.

[image]

---

describe in Japanese

3つのキーを押すと3つのダイオードを通してROW3が L となっても不思議ではない。
ダイオードによる電圧降下を0.7V程度と見做せば、2.6~2.9V程度にはなっていそう。
仮にRP2040で動かした場合は1.1V程度でかなり不安定になりそう。

なお左手側ではLEFT判定のJP9がショートしているので条件を満たしやすい。

koron, 2022-04-17:
> 仮にRP2040で動かした場合は1.1V程度でかなり不安定になりそう。

実際に1N4148使って実験した結果、
1.6VでLo判定で、ちょっとしたノイズでHiにもなりそうだった。

koron, 2022-04-17:
1N4148Wのデータシートによると電流が小さくなると順電圧が小さくなるので、
ハンダ付け不良などにより抵抗が大きくなってLo判定を受ける、という仮説を思いついた。

koron, 2022-04-18:
参考情報: https://blog.ikejima.org/make/keyboard/2019/12/14/keyboard-circuit.html

> 理想的なダイオードは片方に電気を完全に通し、 逆方向には全く通さない。 しかし、現実のダイオードはいくつか理想的ではないふるまいをする。 例えば、ダイオードを図のように接続すると、完全に0Vではなく電位差がある。 良く使われている1N4148はキーボードに使うと最高0.9V低くなる。

これは「最高0.9V低くなる」であって場合によっては「0.5Vしか低くならない場合もある」
このケースにひっかかると0.5x3で1.5Vになり、Lo判定≒ゴーストが発生しうる。
また0.85Vになって2.55Vとなると辛うじてHi判定になるが、
ちょっとしたノイズなどで2.5Vを割ってしまい、ノイズ耐性がないということが起こりうる。

Yowkees, 2022-04-18:
詳細なご考察、ゴースト発生個所に関する図まで本当にありがとうございます。とても分かりやすく理解できました。
また、参考情報リンクも読ませていただきました。ダイオードの電位差のバラつき次第でギリギリなのですね。けっこう危ない橋を渡っていることに気づかされました。
ProMicroが市場からなくなった際にはKeyball61は再設計の必要があると覚悟しました。

koron, 2022-09-10:
rev.2 では回路が修正されたため閉じます。

修正の具体的な内容は、裏のマトリックスとの間でrowかco線のどちらかにダイオードを1個ずつ追加すること。

closed 2022-09-10 by koron