Yowkees/keyball #117: Question: Upstreaming with QMK?

fooker, 2022-04-26:
While hacking on the firmware of my keyball61, I have seen that there are a few components that are self-implemented. Remarkable this is the support for the pmw3360 sensor and the mouse key handling. Is there any reason why these are implemented differently from what upstream QMK provides?

I have managed to build a firmware using the upstream drivers. It looks like it"s working fine. Is there interest in a pull-request for such a change?

In addition I would like to to see support for the keyball boards in upstream QMK. I think that would simplify firware builds and increases visibility of these great keyboards.

Yowkees, 2022-04-26:
Ummm..
I want ask about it Mr.KoRoN.
I think he would find some problem (bug) in PMW3360 upstream drivers.

I hope some day qmk firmware accepts my pull requests.

Any way, thank you for your suggestion. See you on QMK firmware. Thanks!

fooker, 2022-04-26:
Oh, I did not realize that there are already attempts to bring it upstream.

koron, 2022-05-08:
> Is there any reason why these are implemented differently from what upstream QMK provides?

There were two problems on a driver in upstream to use with Keyball.

* The size is too big
    Upstream one includes a frimware for PMW3360DM called SROM.
    But it is not mandatory for PMW3360DM working.

* BUG: can't read registers
    There is a bug to read registers from PMW3360DM.
    Reading registers is required to checking trackball is provided or not.

> Is there interest in a pull-request for such a change?

No thanks.
Current Keyball's driver is working very fine for Keyball.
I don't want to have any extra hassle that doesn't involve Keyball.