Yowkees/keyball #146: Added to upstream

drashna, 2022-10-14:
These really should be PRed to upstream QMK.

If you need any help with that, I'd be more than willing to help out.

Yowkees, 2022-10-25:
OK! I will PR later! Thank you for your help.

drashna, 2022-10-26:
In case it helps, the pointing device code has significantly improved in the last year.  Including support for splits in the core code:

https://docs.qmk.fm/#/feature_pointing_device?id=split-keyboard-configuration
and
https://docs.qmk.fm/#/feature_pointing_device?id=split-keyboard-callbacks-and-functions

There are also a number of keyboards you can look at for reference (bastardkb has the charybdis and dilemma, both of which are split boards with pointing device support).

koron, 2023-01-22:
It looks like there is no time to add it to the upstream.

So let's close this isssue and reopen it when we can take that time.

closed 2023-01-22 by koron