Yowkees/keyball #162: Trackball not working with qmk>= 0.18.0

zhyu, 2022-11-23:
The last qmk version which works well with my keyball39 is 0.17.9.

There are some minor changes needed to compile with qmk 0.18.0, e.g., [RESET to QK_BOOT migration](https://docs.qmk.fm/#/ChangeLog/20220827?id=reset-2-qk_boot), but I still managed to compile keyball firmware with qmk 0.18.0. However, after flashing the firmware, the movement of the trackball is not recognized at all. Maybe it's related to the [pointing device improvements](https://docs.qmk.fm/#/ChangeLog/20220827?id=pointing-device-improvements) mentioned in the changelog.

**Why update?**

There would be a nice feature coming in the next breaking release of qmk that allows automatically activate the mouse layer when the mouse cursor is moved (https://github.com/qmk/qmk_firmware/pull/17962).

Btw, as #126 pointed out, `SAFE_RANGE` has been changed. From my experience, the special keycodes of keyball should be `+1` (e.g., `0x5DA5` -> `0x5DA6`) in qmk 0.16.x. And it should be `+8` (e.g., `0x5DA5` -> `0x5DAD`) in qmk 0.17.x.

zhyu, 2022-12-03:
I have tried to build various commits between qmk 0.17.9 and 0.18.0, and it seems https://github.com/qmk/qmk_firmware/pull/17840 is the change that causing the issue.

zhyu, 2022-12-03:
Confirmed the trackball works fine with qmk 0.18.17 (the last release before 0.19.0) after reverting the commit causing issues.

I tried to build qmk 0.19.0 as well. The trackball still works fine, however, the keymap is no longer recognized by REMAP any more. I guess it could be related to the VIA V3 update: https://github.com/qmk/qmk_firmware/pull/18222

zhyu, 2022-12-03:
Cherrypicked the auto mouse layer feature on qmk 0.18.17, and everything works as expected. So I'm happy with it right now.

If I could find time and fix the trackball issue without reverting the upstream commit, I could send a PR to make keyball firmwares compatible with qmk 0.18.x.

Yowkees, 2022-12-03:
I am sorry I haven't had any action about this issue, because of I am quiet busy lately.
This PR will be helpful to catch up QMK.

Thank you so much for your cooperation;-)

zhyu, 2022-12-04:
Since the issue preventing the trackball from working is about RPC, as a workaround, instead of reverting the corresponding commit, plugging the USB cable to the side with the trackball also works. 

This way, the trackball is on the master side, so no slave-to-master trackball reporting is needed, then no RPC issues :)

closed 2023-01-22 by koron