Yowkees/keyball #169: USBハブ接続時に動作しない

knttnk, 2022-12-23:
USBハブに接続して使おうとしても、すべてのキーとトラックボールが動作しません。

USBハブを使わず、PCに直接繋いだりtypeC-typeA変換ケーブルに繋いだりすると、正常に動作します。
普通のマウスはこのUSBハブに接続しても動作しますので、USBハブの問題ではないと思います。

通常繋いだ方のKeyballにはキーやボールの情報が表示されると思いますが、USBハブに繋ぐとKeyballのロゴが表示されます。この現象と関連がある気がしますが、いかがでしょうか？

### 環境
- Keyball61 rev2
- ファーム: https://remap-keys.app/catalog/RZSU1CrvEW4lns0ww5BM/firmware のKeyball61_via
- USBハブ: https://direct.sanwa.co.jp/ItemPage/400-HUB091S など

koron, 2023-01-22:
こちらで把握している問題としては、
バスパワー(ACアダプタ無し)のUSBハブを用いると電力不足で不安定になるケースがある
というモノがあります。
特にその際はOLEDの表示がされないか不安定になることが知られてします。

ですので、まずはいったんACアダプタを繋いだ状態でも、
引き継き問題が発生するか確認してみてください。

その上で knttnk さんが紹介されている製品 400-HUB091S は、
正確にはUSBハブではなくドッキングステーションであるため、
いわゆる普通のUSBハブとは若干性質が異なっています。
Keyballが利用している、
Pro Microに搭載されているMCUチップ=ATmega32U4や
ベースとなる共通ファームウェア=QMK firmwareが
それらに対応できていない可能性も、わずかながら存在しています。

そのような場合は別の接続手段を取るようにしてください。

knttnk, 2023-01-24:
コメントありがとうございます。

ACアダプタをドッキングステーションに繋いでも、変わりませんでした。

knttnk, 2023-01-24:
masterとslaveの認識がうまく行っていないように見えたので、`qmk_firmware\keyboards\keyball\keyball61\config.h`の
```h config.h
#define SPLIT_USB_DETECT
#define SPLIT_USB_TIMEOUT       500
```
の部分をコメントアウトしてみたところ、使えるようになりました。

もっとうまい解決策はあるでしょうか？これで生じる弊害がなければこのままでもいいのですが。

koron, 2023-09-16:
knttnk さんの環境で `SPLIT_USB_DETECT` を無効化することでうまく動くのであれば
そのようにして使っていただいて大丈夫です。
とくに問題はないと思います。

しかしデフォルトの設定としては現状のままとします。

ProMicroには互換機が何種類もあります。
そのうちの入手が容易ないくつかは `SPLIT_USB_DETECT` が無いと
secondary (slave) 側が動かないという不具合を生じることがわかっています。
(参考: https://zenn.dev/koron/articles/7cdfd5382f9ee3)

そのため、ユーザーが利用しているProMicro互換機を指定できない以上、
この `SPLIT_USB_DETECT` の設定はデフォルトでしておいた方が有利であると
考えて現状維持と決定しました。

あしからずご了承ください。

koron, 2023-09-16:
## まとめ

`SPLIT_USB_DETECT` を設定することで
ドッキングステーション経由でKeyballが認識されない問題が生じることが分かった。

ドッキングステーションにUSB HIDデバイスを接続した(≒Keyballに電源が入った)際に
USBの通信が確率するまでに500msec以上の時間が経過していることが原因だと推定される。

しかし `SPLIT_USB_DETECT` は有効なままとすることにした。
雑多なProMicro互換機の中にはこれがないと正常に機能しないものがあり
(参考: https://zenn.dev/koron/articles/7cdfd5382f9ee3)
現時点においてはそのケースを救うほうを優先した。

closed 2023-09-16 by koron

knttnk, 2023-09-17:
なるほど，丁寧なご対応ありがとうございました．

zhyu, 2023-10-29:
I was facing the same issue.

I understand `SPLIT_USB_DETECT` is required for some ProMicro compatible boards, but probably setting `SPLIT_WATCHDOG_ENABLE` as well is a better default.

By enabling the watchdog, the keyboard will be rebooted automatically if `SPLIT_USB_DETECT` fails, I can confirm it resolves my issue with a USB switch.

Ref:
* https://docs.qmk.fm/#/feature_split_keyboard?id=hardware-configuration-options
* https://github.com/qmk/qmk_firmware/pull/18599