Yowkees/keyball #170: QMKのファームウェアが破壊的なアップデートをして昔のプログラムがエラーになる

metoro0530, 2022-12-31:
http://www.neko.ne.jp/~freewing/hardware/qmk_update_program_with_newest_firmware_2022/
この記事のようなことが起こったので対応していたのですが、最終的に

…\qmk_firmware/keyboards/keyball/lib/keyball/keyball.c:499: undefined reference to `register_button'
 | collect2.exe: error: ld returned 1 exit status
 |
make[1]: *** [builddefs/common_rules.mk:267: .build/keyball_keyball44_default.elf] エラー 1
Make finished with errors
make: *** [Makefile:414: keyball/keyball44:default] エラー 1

というエラーを吐いて停止します。
当該箇所ではこの関数は定義されていないのですが、どのような動作を想定した関数なのでしょうか。
また、どのように修正すればよいでしょうか。

koron, 2023-01-22:
https://github.com/Yowkees/keyball/pull/182#issuecomment-1399414981

> we consider that most of Keyball's users use [Remap](https://remap-keys.app/) to configure keymap.
> And as you may know, current Remap doesn't support QMK 0.19 yet.
> However, they shall plan to shifting to QMK 0.19 ([Japanese post](https://www.eisbahn.jp/yoichiro/2022/12/next_remap.html), and https://github.com/remap-keys/remap/issues/751).
> We will merge this after Remap supported QMK 0.19.

please use QMK 0.16.x (0.16.3 recommended/tested), or use `support-qmk_firmware-0.19-or-above` branch which include #182

metoro0530, 2023-01-22:
Although I somehow managed to use keyball after uninstalling drivers and replacing a USB cable which was too long and could not supply power enough, this advice should be the point.
Thank you.

closed 2023-01-22 by metoro0530