Yowkees/keyball #191: Add pictures of keyboard models

kittydoor, 2023-01-31:
When landing on the repository, there are no images in the README.md, so it is difficult to understand at a glance what the keyboards are like. It would be very helpful to have pictures embedded in the readme.

AlgusDark, 2023-06-01:
This would be nice.

This is an example of a repo that has all the keyboards in an image:
https://github.com/pierrechevalier83/ferris

The readme in Keyball only mentions the 46 (and maybe that one is not manufactured?).

If @Yowkees  has a picture, I could help on creating a README in a PR.

koron, 2023-09-17:
I told this with Yowkees-san, and I was given permission to use photos from the online store and build guides.

I'll take care of this. 👍

koron, 2023-09-17:
I feel Keyball 44 is better.

[image]
[image]
[image]
[image]

koron, 2023-09-17:
I'm going to use Keyball61's photo, it would be fit with OGP image.

koron, 2023-09-17:
#346 is merged, which added top image.

closed 2023-09-17 by koron