Yowkees/keyball #314: Support current version of QMK

sohrab-, 2023-08-20:
The QMK configuration in this repository is implemented using a QMK version from 2-3 years ago. It does not work with the current version of QMK. 

For users that have advanced QMK configuration, they have to compile the firmware from source, instead of REMAP/VIA. This is currently not possible unless we drop down to a much older version of QMK.

Is it possible to update the QMK config to be compatible with the newer versions of QMK? Starting point would be defining a [`info.json`](https://docs.qmk.fm/#/reference_info_json) file. Thanks!

koron, 2023-09-16:
have supported QMK 0.19.10 now. cf https://github.com/Yowkees/keyball/pull/183
i'm going to upgrade QMK 0.22.x

(it is happen by [remap supporting 0.22.2 or higher](https://remap-keys.app/docs/support-qmk-022))

> I believe QMK now [supports](https://docs.qmk.fm/#/feature_pointing_device?id=pmw-3360-and-pmw-3389-sensor) pmw-3360 out of the box.

i know it. but it had many problems with bundled driver.
so i keep to using own one.

koron, 2023-09-16:
Continue to https://github.com/Yowkees/keyball/issues/342

closed 2023-09-16 by koron