Yowkees/keyball #444: Sensor pcb broken

tangbonze, 2023-12-10:
Hi there: I try to build a keyball by myself, found sensor pcb gerber you provide is broken. Could you please kindly check? Thanks a lot.

tangbonze, 2023-12-10:
https://github.com/Yowkees/keyball/blob/main/keyball46/ball_design_data/Trackball_PCB_gerber.zip
here's the file.

koron, 2024-03-24:
FYI @Yowkees

KastenTaco, 2024-04-15:
This issue is still persisting it seems?

tangbonze, 2024-04-22:
[image]
This is gerber file I upload to jlc. You can see it definitely broken.

tangbonze, 2024-05-24:
any update? If this pcb is close source, please just let me know. thanks.

tangbonze, 2024-06-14:
close this by myself since no response at all. If it is close sourced material, you can just say it.

closed 2024-06-14 by tangbonze

ya-luotao, 2024-11-04:
I also encountered the same issue; it seems that the drill layer is missing in the exported Gerber file.