Yowkees/keyball #540: Support for QMK Layouts

sohrab-, 2024-04-03:
As you may know, QMK supports [layouts](https://github.com/qmk/qmk_firmware/blob/master/docs/feature_layouts.md), which allows for the same layout to be applied to multiple keyboards, without the need to configure each keyboard x keymap individually.

Looking at the [default layouts](https://github.com/qmk/qmk_firmware/tree/master/layouts/default), for example, we can in theory apply `LAYOUT_split_3x5_3` to Keyball39.

Having this feature would unlock many of the existing keymapping for keyball keyboards.

I am going to have a look at this myself (although I don't have much QMK knowledge yet) but I thought I would also raise it as an issue if someone has already done the work or wants to collaborate.

koron, 2024-04-05:
> unlock many of the existing keymapping for keyball keyboards

I believe that Keyball loses most of its value without [unique key codes](https://github.com/Yowkees/keyball/blob/main/qmk_firmware/keyboards/keyball/lib/keyball/keycodes.md) in the keymap.
So I don't understand the benefit of being able to use keymaps from other keyboards.

If you know of a specific example of another keyboard keymap that would be meaningful to use with Keyball, please let me know.

sohrab-, 2024-04-06:
I originally wanted this for [Miryoku](https://github.com/manna-harbour/miryoku_qmk/tree/miryoku/users/manna-harbour_miryoku#community-layouts) but I ended up doing that one by adding a new keymap myself.

Here is my `keyball39/keymaps/manna-harbour_miryoku/config.h` for reference:

```c
#pragma once

#define XXX KC_NO

#define LAYOUT_miryoku( \
K00,   K01,   K02,   K03,   K04,          K05,   K06,   K07,   K08,   K09, \
K10,   K11,   K12,   K13,   K14,          K15,   K16,   K17,   K18,   K19, \
K20,   K21,   K22,   K23,   K24,          K25,   K26,   K27,   K28,   K29, \
N30,   N31,   K32,   K33,   K34,          K35,   K36,   K37,   N38,   N39 \
) \
LAYOUT( \
K00,   K01,   K02,   K03,   K04,          K05,   K06,   K07,   K08,   K09, \
K10,   K11,   K12,   K13,   K14,          K15,   K16,   K17,   K18,   K19, \
K20,   K21,   K22,   K23,   K24,          K25,   K26,   K27,   K28,   K29, \
XXX,   XXX,   XXX,   K32,   K33,   K34,   K35,   K36,   XXX \
)

```

(Requires `#define LAYER_STATE_32BIT`.)

erplsf, 2024-12-14:
hi, @sohrab-! could you please share your setup, how you git it compiled for keyball39? I've cheched out and set up latest miryoku, can build it for other keyboards. added your config.h and empty keymap.c but having no luck to getting it compiled.

erplsf, 2024-12-14:
nevermind, I figured it out: https://github.com/erplsf/keyball/blob/main/qmk_firmware/keyboards/keyball/keyball39/keymaps/manna-harbour_miryoku/config.h

619frank, 2025-01-13:
Thank you so much for this @erplsf. finally I was able to install miryoku layout in my keyball39.

I need one more help. do you know how to enable dragscroll on a particular layer?

619frank, 2025-01-17:
I found out how to enable dragscroll. Here is my miryoku custom_config.h

https://github.com/manna-harbour/miryoku/discussions/377