Yowkees/keyball #552: ドキュメント内で「Corneのファームウェア」へのリンクが切れている

sounisi5011, 2024-04-11:
例えば、以下の文章内において、「[Corneのファームウェア](https://github.com/foostan/crkbd/blob/master/doc/firmware_jp.md)」へのリンクが切れています。

https://github.com/Yowkees/keyball/blob/main/keyball61/doc/rev1/buildguide_jp.md

URLが同じであることから、おそらく他のドキュメントでも同様のことが起こっているものと思われます。

https://github.com/search?q=repo%3AYowkees%2Fkeyball%20crkbd&type=code

koron, 2024-04-12:
本来やりたいことはQMK Toolboxの使い方を説明すること。
Corneのドキュメントにリンクすることではない。
現行の[Corneのドキュメント](https://github.com/foostan/crkbd/blob/5cda058e32b16c7600dd50a4d55f43982eb20254/docs/firmware/rev1/firmware_jp.md#flash-the-firmware-by-qmk-toolbox-%E3%82%92%E4%BD%BF%E3%81%86)を見てみたが、
QMK Toolboxの解説として適切とは考えられない。

それを踏まえて検討すると取れる対策は以下のどれか

1. 古いCorneのドキュメントをチェックしたうえで、良さそうならpermalinkでリンクしなおす
2. QMK Toolbox のより良い解説文章を探してリンクする
3. いっそ自分で書いちゃう

koron, 2024-04-12:
> 古いCorneのドキュメントをチェックしたうえで、良さそうならpermalinkでリンクしなおす

* 日本語: https://github.com/foostan/crkbd/blob/fbb146fe372ce8b2b1543ced67367bf19e910945/doc/firmware_jp.md#qmk-toolbox-%E3%81%AE%E3%83%80%E3%82%A6%E3%83%B3%E3%83%AD%E3%83%BC%E3%83%89
* English: https://github.com/foostan/crkbd/blob/fbb146fe372ce8b2b1543ced67367bf19e910945/doc/firmware_en.md#download-qmk-toolbox

悪くはなさそう。とりあえずこちらへのリンク差し替えで対応するべきだろう。

koron, 2024-04-12:
チェックしたところリンクが使われているのは 46 と 61 の合計4つのファイルだけだった。
#553 で修正を用意した。

念のため 39 と 44 も再度確認する。

koron, 2024-04-12:
みてみたところ Remap のリンクを案内しているようだった。

例: https://remap-keys.app/catalog/tAJ9Htme4oNabUkx4832/firmware

46は良いとして、61のほうは書き換えたほうが良いかもしれない。

koron, 2024-04-12:
61のドキュメントを見たら、英語版からは当該リンクを含む `2-2 ファームウェア` がなくなってる。
というか最初の commit からなかった。
消しちゃって良さそう。

koron, 2024-04-12:
39と44の日本語版には別の書き方がされてそう。それにあわせるか。

koron, 2024-04-12:
## まとめ

QMK Toolboxの説明に関して、切れていたリンクは4ファイル4か所でした。
その内、現状販売されていないKeyball 46の2ファイル2か所についてはpermalinkを用いてリンクを書き換えました。

現行機種であるKeyball 61については
同じく現行機種であるKeyball 39 や Keyball 44 のドキュメントに合わせて
REMAPへと案内するように文言の修正をした結果、
該当リンクを削除することになりました。

変更の詳細は #553 を参考にしてください。

closed 2024-04-12 by koron