Yowkees/keyball #566: 【keyball39】QMK firmwareでビルドしたhexファイルを書き込むとキーマップが反転する

TakumiMurai, 2024-04-24:
### 背景
オートマウスレイヤーやOLEDをカスタマイズしたく、QMK firmwareの環境構築を行い公式のファームウェアをビルドして動作確認しようとしたところ、キーマップやポインタの動きが反転してしまいました(裏返して使っているようなイメージ)

### やったこと

1. QMK firmwareのインストール
2. keyballリポジトリのクローン
3. `make keyball/keyball39:via`コマンドの実行
4. pro micro web updaterを用いて書き込み

### 環境
QMK firmware ver. 0.24.7
windows 11

### その他
公式で案内のあったremap上のファームウェア書き込みでは正しく動作しております。
逆転症状が発生している状態でremapを確認すると、正しいキーマップが表示されます。

何卒よろしくお願いいたします。

okuRaku, 2024-04-24:
Keyball relies on an older version of QMK than 0.24.7.  I noticed the same behavior you did when I tried to use newer QMK, however other features will also break I believe.   Try 0.22.14 instead.   How to check out this specific version for your local build environment is detailed [here](https://github.com/Yowkees/keyball/tree/main/qmk_firmware/keyboards/keyball) in step 2 (basically, check out the versioned tag)

TakumiMurai, 2024-04-24:
Thanks! With 0.22.14, built firmware worked correctly.
I didn't know there is official firmware build guide.

I will close this issue.