Yowkees/keyball #572: Keyball61 on rp2040

idank, 2024-05-02:
Hi,

I previously had success with Keyball44 conversion to rp2040, and now I tried to convert Keyball61 and run into some issues. I tried to disable all features and just test the keyboard matrix and got stuck there where no key presses are registered. I noticed Keyball61 uses custom matrix scanning code called "duplexmatrix" and this is my first suspect where the compatibility issue might be. @koron I see you write this code, any idea why this wouldn't work on rp2040? To rule out soldering issue, I tested the keyboard with pro micro and it is working.

Here are the changes I made to the firmware to compile it for rp2040: https://github.com/idank/qmk_firmware/commit/ef9a3dc0d333118c46261c93e8077ac369039972

The same repo has working rp2040 Keyball44 firmware.

idank, 2024-05-05:
I managed to solve it by modernizing the matrix scanning code, see https://github.com/idank/qmk_firmware/commit/c78580f0a4dac67c98d7bfb61728288ecc524004.

closed 2024-05-05 by idank

carlos-gtz, 2024-06-06:
@idank , what are the changes needed for the keyball44 to work with rp2040?

idank, 2024-06-06:
You can find a working firmware on my clone https://github.com/idank/qmk_firmware/tree/dev-rp2040/keyboards/keyball

ljmill, 2024-06-11:
@idank What all would be required to be changed to get the RP2040 to work with a Keyball39?