Yowkees/keyball #588: Issue about updating to qmk_firmware 0.24.11

cyrusn, 2024-05-19:
I am attempting to compile and flash firmware version 0.24.11 to my Keyball61 keyboard. Despite encountering warnings, I proceeded with flashing the hex file. After flashing, I noticed that the left and right keyboards were swapped, and the trackball movement was inverted.

``` sh
☒ keyball/keyball61: LAYOUT_right_ball: Layout macro should not be defined within ".h" files.
☒ keyball/keyball61: LAYOUT_left_ball: Layout macro should not be defined within ".h" files.
☒ keyball/keyball61: LAYOUT_dual_ball: Layout macro should not be defined within ".h" files.
☒ keyball/keyball61: LAYOUT_no_ball: Layout macro should not be defined within ".h" files.
```

CreamNuts, 2024-06-10:
My problem is same as yours. Do you have any solutions?

cyrusn, 2024-06-11:
No idea too.

CreamNuts, 2024-06-12:
I found it. Change `#define SPLIT_HAND_MATRIX_GRID` to `#define SPLIT_HAND_MATRIX_GRID_LOW_IS_LEFT` in `keyball61/config.h`. Note that `keyball61/config.h` is different from `keymaps/custom/config.h`

cyrusn, 2024-06-12:
> I found it. Change `#define SPLIT_HAND_MATRIX_GRID` to `#define SPLIT_HAND_MATRIX_GRID_LOW_IS_LEFT` in `keyball61/config.h`. Note that `keyball61/config.h` is different from `keymaps/custom/config.h`

Oh ... sadly it didn't work for me.

XenotropicDev, 2024-10-12:
Worked for me too. Make sure you add another line for it and don't replace the existing SPLINT_HAND_MATRIX_GRID.
Should look something like this.

```c++
#define SPLIT_HAND_MATRIX_GRID  F7, D7
#define SPLIT_HAND_MATRIX_GRID_LOW_IS_LEFT
```

divad1196, 2025-10-25:
Hi, The `#define SPLIT_HAND_MATRIX_GRID_LOW_IS_LEFT` did fix the left/right issue.
The trackball behavior had to be fixed in the `keyball_on_apply_motion_to_mouse_move` function.

```C
__attribute__((weak)) void keyball_on_apply_motion_to_mouse_move(keyball_motion_t *m, report_mouse_t *r, bool is_left) {
// #if KEYBALL_MODEL == 61 || KEYBALL_MODEL == 39 || KEYBALL_MODEL == 147 || KEYBALL_MODEL == 44
//     r->x = clip2int8(m->x);
//     r->y = clip2int8(m->y);
//     // r->x = clip2int8(m->y);
//     // r->y = clip2int8(m->x);
//     if (is_left) {
//         r->x = -r->x;
//         r->y = -r->y;
//     }
// #elif KEYBALL_MODEL == 46
//     r->x = clip2int8(m->y);
//     r->y = -clip2int8(m->x);
// #else
// #    error("unknown Keyball model")
// #endif
    r->x = clip2int8(m->y);
    r->y = clip2int8(m->x);

    // clear motion
    m->x = 0;
    m->y = 0;

}
```

It's a quick and dirty fix, we can just keep all the macro definition. The important this is to cross assign `m->y` to `r->x`.