Yowkees/keyball #596: Issue about AUTO_MOUSE_THRESHOLD

o-matsu, 2024-05-30:
I added `AUTO_MOUSE_THRESHOLD` to `qmk_firmware/keyboards/keyball/keyball39/keymaps/{mymap}/config.h` to customize Auto Mouse Layer.
The optional parameters were taken from qmk's site. https://github.com/qmk/qmk_firmware/blob/master/docs/feature_pointing_device.md#how-to-customize

However, after building and writing to the device, the behavior does not change.
Is this option not supported by the keyball firmware?

okuRaku, 2024-05-30:
Hello @o-matsu -san.   I noticed the same and confirmed it is in a newer version of the firmware than is in use for keyball.   I adapted the code to our version and put a pull request here:  https://github.com/Yowkees/keyball/pull/563   

Maybe it will help you, I hope so!

o-matsu, 2024-05-30:
Thank you @okuRaku  -san. I'll try it.

o-matsu, 2024-05-31:
Hi, @okuRaku -san.
Thank you for your kind advice.
I checked out your branch and tried it.
At first glance, it seemed to work well, but there were some strange behaviors.
Specifically, after the mouse layer was activated by moving the cursor, the layer was deactivated even though I kept moving the cursor.

As a provisional measure, I thought the approach of independently implementing the modifications to the QMK firmware was a good idea.
I will also try to see if I can implement it successfully.
Thank you.

okuRaku, 2024-05-31:
I look forward to your changes!