Yowkees/keyball #636: トラックボールの一時的速度変更について

masatosuzuta, 2024-07-28:
先日Keyball 39を購入させていただき、試行錯誤しながら大変便利に使わさせていただいてます。

1つ質問なのですが、特定のキーを押している間のみトラックボールのCPIを上げるまたは下げることは可能でしょうか？

私がトラックボールの使用が初めてなのと、アイコンの小さいソフトウェアを使うことが多く、調度良いCPIがなかなか見つからず、精密な作業用の遅いCPIと通常のカーソル移動用の速いCPIの2つを切り替えられたら便利だと考えた次第です。

お忙しい中恐縮ですが、ご教授いただければ幸いです。よろしくお願い致します。

yoichi, 2024-10-14:
例えばkeymap.cで以下のようにすると、Kb 18を押している間だけCPI=200になり、離すと元に戻ります。

```
bool process_record_user(uint16_t keycode, keyrecord_t *record) {
    static uint8_t prev_cpi_value;
    switch (keycode) {
    case QK_KB_18:
        if (record->event.pressed) {
            prev_cpi_value = keyball_get_cpi();
            keyball_set_cpi(2); // 一時的に使うCPIを100で割った値を指定
        } else {
            keyball_set_cpi(prev_cpi_value);
        }
        return false;
    default:
        break;
    }
    return true;
}

```
　

（keyball_set_cpi()のヘッダコメントが間違っていたので後でPull Request出しておきます）

masatosuzuta, 2024-10-15:
yoichi様

大変有難うございます！早速試してみます！