Yowkees/keyball #648: ProMicro Data端子とtrrsジャック間が導通していない

takeru1968, 2024-08-12:
質問失礼致します。
当方、初めての自作キーボード作成のため、
情報不足等至らない点があるかと思われますが、
ご教授して頂けると幸いです。
## 概要
左手側基盤(非トラックボール側)ProMicroのData端子とtrrsジャックData端子が導通していない
## 詳細
- 下記issueと同様の症状が発生(USB接続とは反対側のキーがすべて反応しない)
[TRSケーブルで繋いだ方のキーボードが認識されない #245](https://github.com/Yowkees/keyball/discussions/245)
[片方のキーボードが認識されない #301](https://github.com/Yowkees/keyball/discussions/301)
- issue同様に左右ProMicroのData端子をテスターで導通確認したところ、導通していない。
- さらに細かく調べたところ、右手側ProMicroのData端子から左手側trrsジャックまでは導通していることを確認。
- 左手側ProMicroのData端子とGNDは導通(画像赤丸部をテスターでテスト)
[image]
- 左手側のProMicroとtrrsジャックの端子が導通していない(テスターで画像の赤丸同士をテスト)
[image]
[image]
[image]
[image]
- 上記の場合の対応策、追加で確認が必要な箇所等がある場合はお教えしていただきたいです。

以上、ご回答のほど、よろしくお願い致します。

Yowkees, 2024-10-02:
投稿に気付けずご回答がとても遅くなり申し訳ありません。
すでに解決済かも知れませんがご回答させていただきます。

下記画像の緑丸部のパターンが剥がれてDATA線が断線していると思われます。
そのため、青点どうしをエナメル線等で接続していただくと解決します。
お手数ですがご対応お願いいたします。
[image]

takeru1968, 2025-02-26:
> 投稿に気付けずご回答がとても遅くなり申し訳ありません。 すでに解決済かも知れませんがご回答させていただきます。
> 
> 下記画像の緑丸部のパターンが剥がれてDATA線が断線していると思われます。 そのため、青点どうしをエナメル線等で接続していただくと解決します。 お手数ですがご対応お願いいたします。 [image]

返信いただき、ありがとうございます。
回答の期間の方が空いてしまい申し訳ありません。
上記青い点を銅線ではんだ付けの上、
導通を確認したのですが、
症状が改善していない状態となっています。
追加で確認が必要な箇所がある場合、
ご教授頂ければと思います。

[image]