Yowkees/keyball #677: ジャンパの基盤を剥離させてしまった。トラックボールが反応しない。keyball39

Dr-cafelatte, 2024-10-14:
先日keyball39を購入させていただきました。
右手トラックボール側の光学センサー用信号線のジャンパ形成が上手くいかず、基盤が剥離してしまいました。
お手数ではございますが対処法があれば教えていただけますでしょうか。
[image]
[image]

Dr-cafelatte, 2024-10-15:
追記。
はんだが得意な人と確認し、画像のように右手トラックボール側表面を、クロスを作るようにジャンパを形成しましたが、変わらずOLEDの移動量も全て0のままでした。
[image]

Yowkees, 2024-10-29:
ご連絡が遅くなり申し訳ありません。

修理の方法は適切です。
下記の同色同士が導通しているか、また違う色どうしがショートしてしまっていないかご確認をお願いします。

[image]

Dr-cafelatte, 2024-11-21:
違う色同士のショートは確認できませんでした。
また、青色同士のみ導通しておりませんでした。

Yowkees, 2024-11-22:
ご確認ありがとうございます。
青点どうしが導通していなかったということで、下記画像の青点どうしをエナメル線か導線で接続してください。
これでトラックボール復活できると思います。
[image]

Dr-cafelatte, 2024-11-22:
ありがとうございます！
無事トラックボールが反応するようになりました。
部品を買い替えることなく修理できると思っていなかったので本当に感謝しております。
忙しい中、迅速な対応ありがとうございました。

closed 2024-11-22 by Dr-cafelatte