Yowkees/keyball #684: 特定のキーが勝手に反応しない&特定のキーが勝手に反応してしまう

tatatata222, 2024-10-29:
初めての自作キーボードでトラブルが発生してしまい困っています。

どなたかご教授いただけますと幸いです。

現在keyball39を使用しております。

現在起きている問題は下記のとおりです。

▶︎レイヤー1が勝手に反応してしまう

現在レイヤー1にテンキーを仕込んでいます。

ただタイピングをしている時に勝手にレイヤー1になってしまい、文字入力をする時に数字をご入力してしまいます。

キー側の問題なのかな？と思いつつ現在仕込んでいる親指の一番左側からキーの場所を変更し直したり、半田付けをしなおしたりしてみたのですが、症状が改善しません。

▶︎レイヤー2と3が反応しない時がある

上記の問題に付随してなのですが今度はレイヤーが反応しない時があります。

これもキーや半田付けのミスによるものなのかな？と思いつつキーを変更しても症状が改善しませんでした。

接触不良が問題ではなさそうなので、膨大な労力をかけて作成したにも関わらずお手上げ状態です。。。

現在もこちらの文章はkeyballを使っているので我慢すれば使えるっちゃ使えるので巣が困っています。

どなたかお答えいただけますと幸いです。

tatatata222, 2024-10-29:
またこちらのキーボールですが使っているマックブックの電源を落とすと再度接続しない押さないと接続がされないのと、何度の接続をしなおしても接続がされない時があります。

一度つながると接続不良になることはないので接触自体に問題はないとは思うのですが、そこそこ不便で困っています。この点に関してもお答えいただけますと幸いです。

tatatata222, 2024-10-29:
追加です。レイヤー以外に他のキーが勝手に押される現象はございません。

Yowkees, 2024-10-30:
ご連絡ありがとうございます。

ちょっとこれまで聞いたことがない現象です。
レイヤー切り替えボタンを他のキーに移動しても再現しますか？
キーマップのスクリーンショットを共有いただければ幸いです。

以上、お手数ですがご確認お願いいたします。

tatatata222, 2024-10-30:
お世話になります。

レイヤー切り替えボタンを移動しても再現されました。

ただファームウェアの方を最新に更新したら不具合が完全に解消しました。

早急にご対応いただき感謝申し上げます。

引き続きどうぞよろしくお願い申し上げます。