Yowkees/keyball #695: スクロール停止直後にcursorが, ピクピク動く現象が発生

kyo219, 2024-11-18:
現象を説明するために動画を添付します. 
(説明性のため, 感度をあえて高くしています) 
https://github.com/user-attachments/assets/6cd648a3-709f-4034-a897-d821f863d7ee

同様の, 現象が起きている方はいらっしゃいますでしょうか? 
現在, 黒色のトラックボールを使用していますが, 他色の方が読み取り精度が良い等あればご教示いただきたいです.

kyo219, 2024-11-21:
トラックボール台がブカブカなことが問題な気がしてきました. 
現状ネジをしっかり閉めた状態ですが, 机の揺れ等でもトラックボールが揺れます（それによりcursorがピクピクする.

Yowkees, 2024-11-21:
ご連絡ありがとうございます。

トラックボールが不安定ですか？
ブカブカというか、ボールはプラスチックケースには触らず、セラミックボール３個で支持されています。
私の手持ちのKeyballではキーボードを揺すってもボールやカーソルは動きません。
もしかするとセラミックボールが無くなっている、もしくはセラミックボール以外にボールに触れているものが無いかご確認お願いします。

kyo219, 2024-11-22:
@Yowkees 

ご回答ありがとうございます.! 
こちら確認したところ, セラミックボールが無くなっておりました.. 

以下のような, 2mmタイプのセラミックボールを3つ購入すれば代替となりえますでしょうか？
https://www.monotaro.com/g/01218880/

kyo219, 2024-12-12:
こちら、2mmタイプのセラミックボールを追加購入して、挿入したところ改善されました。
従ってclose

closed 2024-12-12 by kyo219