Yowkees/keyball #726: qmk_firmwareのビルドが、`RGBLIGHT_LED_COUNT`という変数に関するエラーで成功しない

taichi-k, 2025-01-02:
（注：私は、自作キーボード・qmk_firmwareに関して、全くの素人です。的外れな起票をしていたら申し訳ございません。）

https://note.com/yinouet1001/n/n856b45220ad4
に従い、mac環境にて、qmkを利用したkeyballのfirmwareのビルドを試みたのですが、以下のエラーがでて`.hex`ファイルの生成ができませんでした。

```
QMK Firmware 0.27.4
⚠ keyball/keyball39: Build marker "keyboard.json" not found.
☒ keyball/keyball39: LAYOUT_right_ball: Layout macro should not be defined within ".h" files.
☒ keyball/keyball39: LAYOUT_left_ball: Layout macro should not be defined within ".h" files.
☒ keyball/keyball39: LAYOUT_dual_ball: Layout macro should not be defined within ".h" files.
☒ keyball/keyball39: LAYOUT_no_ball: Layout macro should not be defined within ".h" files.
Making keyball/keyball39 with keymap default

⚠ keyball/keyball39: Build marker "keyboard.json" not found.
☒ keyball/keyball39: LAYOUT_right_ball: Layout macro should not be defined within ".h" files.
☒ keyball/keyball39: LAYOUT_left_ball: Layout macro should not be defined within ".h" files.
☒ keyball/keyball39: LAYOUT_dual_ball: Layout macro should not be defined within ".h" files.
☒ keyball/keyball39: LAYOUT_no_ball: Layout macro should not be defined within ".h" files.
avr-gcc (Homebrew AVR GCC 8.5.0_2) 8.5.0
Copyright (C) 2018 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Compiling: platforms/avr/drivers/ws2812_bitbang.c                                                  In file included from platforms/avr/drivers/ws2812_bitbang.c:26:
drivers/ws2812.h:60:30: error: 'RGBLIGHT_LED_COUNT' undeclared here (not in a function); did you mean 'RGBLIGHT_SAT_STEP'?
 #    define WS2812_LED_COUNT RGBLIGHT_LED_COUNT
                              ^~~~~~~~~~~~~~~~~~
platforms/avr/drivers/ws2812_bitbang.c:154:26: note: in expansion of macro 'WS2812_LED_COUNT'
 ws2812_led_t ws2812_leds[WS2812_LED_COUNT];
                          ^~~~~~~~~~~~~~~~
 [ERRORS]
 |
 |
 |
make[1]: *** [.build/obj_keyball_keyball39_default/ws2812_bitbang.o] Error 1
make: *** [keyball/keyball39:default] Error 1
Make finished with errors
```

mac環境以外にも、Linux環境でも同様のエラーが発生してしまったため、起票に至ります。

toxaO, 2025-01-14:
qmkのバージョンの問題だと思われます。
0.27系で試したところ同様のエラーが出ました。
0.24系でコンパイルすると通ると思われます。
根本的な原因に関しては調べたのですが解決しませんでした……

yoichi, 2025-03-19:
参照されている手順
https://note.com/yinouet1001/n/n856b45220ad4
が、qmk/qmk_firmware の最新をcloneする手順のようですが、keyballの公式のファームウェアビルド手順
https://github.com/Yowkees/keyball/tree/main/qmk_firmware/keyboards/keyball#how-to-build
に従ってQMK 0.22.14を使うようにする
```
$ git clone https://github.com/qmk/qmk_firmware.git --depth 1 --recurse-submodules --shallow-submodules -b 0.22.14 qmk
```
とどうでしょうか？

taichi-k, 2025-06-23:
@toxaO @yoichi 
コメントいただきありがとうございます！
教えていただいた通り、qmk_firmwareのバージョンによる問題の可能性が高そうですね...

諸事情により、私の手元では問題解決を試せなくなっているのですが、
👍 リアクションが、toxaO様のコメントについていることから、バージョンを揃えることで解決できそうかなと思います。

一定期間コメントが付かなければ、解決としてCloseさせていただきます。
助かりました🙇 ありがとうございました。