Yowkees/keyball #749: Trackball PCB kicad files

tirorex, 2025-02-05:
Thank you so much for a project!

Couldn't find KiCad files for Trackball PCB (where mount sensor). 

Please, add it to project. 

Thank you in advance.

talex-tnt, 2025-04-18:
Hello there, I couldn't find them either,
It would be great to have them as well.

Thanks the whole project!

talex-tnt, 2025-04-18:
Well, I managed to find some files here: https://github.com/Yowkees/keyball/tree/main/keyball46/ball_design_data

tirorex, 2025-04-18:
> Well, I managed to find some files here: https://github.com/Yowkees/keyball/tree/main/keyball46/ball_design_data

These are gerbers,  unusable for production (missing drill and other layers). 

Need to either fix it or open KiCad files

talex-tnt, 2025-10-07:
Found one on AliExpr for 20/30$ and replaced the busted one.. it was not identical but worked just fine.