Yowkees/keyball #766: Can't disable RGB in firmware

tbsmn, 2025-03-21:
When I try to disable RGB lightning to safe some firmware size by setting `RGBLIGHT_ENABLE = no` in `qmk_firmware/keyboards/keyball/keyball44/keymaps/custom/rules.mk` of my keymap, the firmware compiles correctly but  the scrollball is not available anymore.

Any idea what I am doing wrong?

yoichi, 2025-04-21:
Didn't you comment out the entire layer_state_set_user()?

yoichi, 2025-04-21:
Do you have layer_state_set_user() in your keymap.c?

Following code in default/keymap.c provides scroll feature:
```
layer_state_t layer_state_set_user(layer_state_t state) {
    // Auto enable scroll mode when the highest layer is 3
    keyball_set_scroll_mode(get_highest_layer(state) == 3);
    return state;
}
```

tbsmn, 2025-04-22:
Yes I have this code in my keymap.c. Sorry I guess I have been a littlebet unspecific. 
It is not only the scrollfunction, but the whole trackball that is missing. I cant even move the mouse.

yoichi, 2025-04-24:
If you change only RGBLIGHT_ENABLE = yes in the situation where the problem occurs, will the trackball be usable?
This is to check if there are other differences causing the problem or if it is a hardware problem.

trsxxii, 2025-08-30:
@yoichi @tbsmn 
Is there any update regarding this issue?
Setting RGBLIGHT_ENABLE=no causes the trackball to stop responding, too.

yoichi, 2025-08-30:
@tbsmn @trsxxii 
When I received the report, I was unable to reproduce it.
However, I just found a way to reproduce the problem. Is this the same as your situation?

How to reproduce:
1. Write the firmware built with RGBLIGHT_ENABLE=yes to the ProMicro on the board with the ball (e.g. Keyball44_via_v1.3.2 in https://remap-keys.app/catalog/tAJ9Htme4oNabUkx4832/firmware).
2. Write the firmware built with RGBLIGHT_ENABLE=no in rules.mk to the ProMicro on the board without the ball.
3. Connect the ProMicro on the board without the ball to your PC.

Solution:
Flash the same firmware onto the ProMicro on both boards.
If different firmware is flashed, they will not function properly.

trsxxii, 2025-08-30:
@yoichi 
Thank you so much. It's resolved. 
I had forgotten that other settings usually work even with firmware written to only one side.

yoichi, 2025-09-21:
@tbsmn I'm glad the problem was resolved. Could you please close this issue?

closed 2025-09-22 by tbsmn